bres55/Smart-BMS-arduino-Reader #1: Problem in current reading

leofalp, 2018-08-21:
I'm trying to red the current value from a "smart BMS" with arduino but using your code the current value is alwais above 655 (mA ?). This is not correct (i'M testing current with other external instruments). If i read the value through the PC with the "official" program the current value is correct.
Maybe we're reading the wrong register?

leofalp, 2018-08-21:
This is the bytes read from serial:
`15	133	0	0	8	232	11	158	0	1	36	227	0	0	0	0	0	0	32	77	3	10	2	11	114	11	131	251	41	

15	123	255	147	8	232	11	158	0	1	36	227	0	0	0	0	0	0	32	77	3	10	2	11	114	11	131	249	161	`

both line are response to che request of basic infos. 
The fisrt without a load: current in 0 beause the third and fourth byte are 0. 
The second line is with a load: third byte goes to 255 and second to 147 regardless of the load.
This is very strange because all the other value are perfect... 
Someone have some ideas?

bres55, 2018-09-23:
You were right, Current miss reading, used unsigned, needs to be signed, current going in(+), or current going out(-)
Hope this sorts it, I have not been able with my set up, to check the going in (charge) current, but the going out (discharge current) is working for me now.
Let me know if you are able to check this for your set up.

bres55, 2019-05-20:
I can confirm now the issue was "float PackCurrentf", so value is positive or negative, depending on power going in or coming out from batteries, via BMS.

closed 2019-05-20 by bres55