bres55/Smart-BMS-arduino-Reader #2: BMS no longer reads with JBDTools

ckbliss49, 2019-05-13:
After running arduino reader, the JBDTools no longer works.  It looks like it is reading but the data screen remains blank.  It feels like the board has gone to sleep and I have not found any way of waking it up.  Is there some message I need to send to reset it to work with JBDTools?  If I run the Arduinoi reader code the BMS responds properly with the Arduino but fails with JBDTools.

bres55, 2019-05-20:
Can't really talk about the JBDTools. But if it works with, Smart-BMS-arduino-Reader, I would suggest checking the com port in the JBDT software, baud rates etc.

ckbliss49, 2019-05-20:
JBD works on other BMS boards. I have not tried Reader with other BMS boards because I do not want to risk having the same problem with them.

On the one I had hooked up and works with Reader, I see TX LED on FTDI box but no RX.  It acts like BMS is in a sleep state. I don't know if relavnet, but PC serial box only has 3 wires(Tx Rx Gnd) whereas Arduino has 4 wires (Tx Rx Gnd +5).  The BMS is a 14 cell unit and the lowest cell is at 3.5 volts.
Perhaps when more boards arrive, I will try on one of them.

bres55, 2019-05-20:
I have swapped between arduino reader and JBDLTools, with no problems, but I use this UART 

https://www.ebay.co.uk/itm/6Pin-USB-2-0-to-TTL-UART-Module-Converter-CP2102-STC-Replace-FT232-CF/272249732398?epid=503069058&hash=item3f63593d2e:g:QVUAAOSw71BXP92B

between my PC, running JBDLTools, and the BMS. I dont have an FTDI box.

Your problem sounds like a hardware, protocol issue.
Not sure what to suggest, But understand reticence, not wanting to put other boards at risk.
If you find what it is, do let me know.

ckbliss49, 2019-05-22:
I  think I have it solved.  I tried a direct serial connection which didn't work then tried a different FTDI board that had a 5v line (VCC). That worked, so I am going to assume that the 5V from the Arduino damaged something on the BMS board.  When the new BMS shows up, I will try connecting it without the +5v from the arduino and see if that solves the problem.  Note: the PC box from the manufacturer only has 3 wires, tx, rx and Gnd.

bres55, 2019-05-22:
That's great news. I think you are right about not having the 5v from arduino to BMS. I have had it connected, but at the moment it is not, and functioning fine. I think from the BMS, it is a 5v output supply, it is certainly needed by the bluetooth module. From observations only, I believe the BMS gets its power from the batteries it's connected to, and I believe, it only takes its power from cells 1 and 2, but I haven't followed up testing that theory. I also put a 100K pull up resistor on each of the RX and TX lines, to improve reliability, but I do have it connect via, a 10 Meter UTP cable! 
It's a while since I put all this up here, but will change the readme, to not include the 5V. Thanks for upadate.
Best of luck with the project.

ckbliss49, 2019-05-25:
On a new SmartBMS I connected the Arduino without the +5volts and it worked fine.  I then connected using JBDTools with their FTDI PC box and finally using their BlueTooth module.  All worked and I can swap between them. I am convinced that connecting the +5 from the Arduino fried something on the oringinl BMS, which is now not working at all.

Another problem I had was reading the first two cells low.  Upon writing <james@lithiumbatterypcb.com> , I got a response that not hooking up B- to the B- point on the board would cause that.  I attached the B- pack wire to the BMS B- and that solved that problem.

bres55, 2020-01-23:
have changed the readme, to not include the 5V

closed 2020-01-23 by bres55

boboxx, 2020-03-11:
On mime (JBD-SP04S020-L4S-12A-B-U) the VDD is 12v!