bres55/Smart-BMS-arduino-Reader #3: possible to use with 10s version?

orhanyor, 2019-07-31:
i checked out the supplier and he has bunch of other options too. is it possible to use this code with 10s version of the bms?

orhanyor, 2019-07-31:
this must be the 10s version. https://www.aliexpress.com/item/32886356039.html?spm=2114.12010612.8148356.3.46637ae46T4WGW

ckbliss49, 2019-08-01:
I have a 11s and a 14s, it works with both.

ckbliss49, 2019-08-01:
DO NOT hook up the 5V from Arduino to the serial Vcc pin on the BMS.  It will fry it.  I learned that the hard way.

orhanyor, 2019-08-01:
oh ok thanks for the tip, so i should connect only TX/RX and GND pins? also my arduino is 3.3v logic i guess bms serial pins are 5v right? then i have to use logic converter too or may be just for the tx pin of the bms?

bres55, 2020-01-23:
based on comments from ckbliss49, I will assume this works with 10s version of the bms

closed 2020-01-23 by bres55