bres55/Smart-BMS-arduino-Reader #4: Modified Version

ckbliss49, 2019-10-01:
I have been working on modifications and wonder about how to post.  My goal is to use the code as a basis for outputting to a 4DSystems display,  So far, I have modified functions to support showing in Notepad++ function list as there is a Notepad++ feature(bug) that prevents that. In actual code, I have only added a global for the cell count.  I am thinking about changing how data is stored to simplify how I output to the new display.   Any other suggestions happily accepted.

ckbliss49, 2019-10-07:
I modified the code to support all  32 possible cells (was 8) for balance display and line up for cell values.  I will upload if anyone needs it, but for now am continuing on patching.

flashtel, 2019-12-24:
Hi ckbliss49, I'm running 2x 20S versions. I am definitely interested in the modification you speak of. I just checked out your profile, can you fork the code to your account and add the modifications there?

bres55, 2020-01-23:
This will be for ckbliss49 to impliment if they wish.

closed 2020-01-23 by bres55