bres55/Smart-BMS-arduino-Reader #5: Discharge current is wild

bkenn011, 2019-10-09:
Using this code, if I'm putting 14.9 amps in to the battery I can see that value appropriately, 14.9.
If I begin discharging at any amount, I get a number in the 6500 range that doesn't seem to respond appropriately whether my discharge goes from 0.1a to 10a.

Any idea what's up?  This is a BMS with shared charge and discharge port FWIW

ckbliss49, 2019-10-12:
Is PackCurrent defied as an int or an unsigned int?  Perhaps it is an issues with negative numbers.  I am hoping to set up some kind of load so I can verify output of a load. Observation that the max for an signed int is 65,535.  Looks a little like the 6500 you show.

ckbliss49, 2019-10-12:
I hooked up a 100 watt light bulb and the current correctly reads -0.75 amps.  I changed the data type for  PackCurrent from int to uint16_t and the current now reads a huge number like 654.32,

26Zoll, 2019-11-25:
With a few additional lines it is working correct now.

  // CURRENT
  highbyte = (inInts[2]); // bytes 2 and 3
  lowbyte = (inInts[3]);
  int PackCurrent = two_ints_into16(highbyte, lowbyte);
  float PackCurrentf;
  if (PackCurrent <= 32768) {
    PackCurrentf = PackCurrent / 100.0f; // convert to float and leave at 2 dec places 
  }
  else {
    PackCurrentf = ((65536 - PackCurrent) / 100.0f) * -1; 
  }
  Serial.print("   Current = ");
  Serial.print(PackCurrentf);

bres55, 2020-01-23:
This was dealt with here
https://github.com/bres55/Smart-BMS-arduino-Reader/issues/1

closed 2020-01-23 by bres55