bres55/Smart-BMS-arduino-Reader #7: Active BMS using arduino

Alpa64, 2020-01-18:
Hello bres55, I really appreciated your post, great job. What do you think of using this device: https://it.aliexpress.com/item/4000191884856.html?spm=a2g0s.9042311.0.0.47564c4dgXFCyP connected to the entire battery voltage and used to charge the weak cell by processing the incoming data from the smart LLT BMS via bluetooth? It is isolated and could be turned on by the arduino board with a mosfet with the source on the ground and the drain on the isolated converter. For each cell an isolated converter to make the balance active up to a current of 1A. Energy is taken from the entire battery pack and sent to the weak cell (s)

bres55, 2020-01-23:
I don't think that will work. The problem is ...Common Ground..., if you try to charge a cell or set of cells in parallel, halfway up the series, then there will be problems. 
A couple of years ago, when I started to build my system, I read about problems, balancing cells, in series, and possible problems. So I looked at options and went for the smart BMS, I spent a long time hacking into it so I could just read the cell voltages, and so, with isolating relays, paralled discharge balancing, ie if the BMS was balancing, I would also balance, I was using 40ma fans.
However, having spent all that time and money, afraid my batteries would not be good enough, I have never needed to use my suplimental balancing.  Apart from when being cahrged, or heavely discahrged, the cells stay within reasonable limits. Delta, between 20 and 60 mv. Usually around 30mv.
Good luck with your project.

Alpa64, 2020-01-23:
Thanks, I agree with you and I am experimenting with it in my powerwall too, I have not yet done many charge and discharge cycles but all groups of cells are well within 20mV even discharging the battery up to about 4500W. The BMS has 50mA of balancing current and I saw its intervention only when I set the intervention threshold to 20mV.

closed 2020-01-23 by Alpa64