bres55/Smart-BMS-arduino-Reader #8: voltage on TX pin of BMS?

orhanyor, 2020-03-13:
Hey guys im planning to use this bms with a 3.3V arduino and pins are not 5V tolerant. So, receiving 5V from BMS TX pin will probably kill the RX pin of the arduino. Can you please tell me the voltage of the TX pin on the bms so i can decide if i need a logic level converter or not thank you.

lorneb, 2020-04-07:
Its 5V,  I eventually found this out after a few hours of mucking around. What I tried first was a USB - TLL adapter set at 3.3V. nothing,  Had to change it to 5v then it worked no problem.

Don't use the 5V from the BMS unless you really have to, just connect tx,rx, gnd
I'm you put a meter across the BMS rx, and you should read 5v

orhanyor, 2020-04-17:
> Its 5V, I eventually found this out after a few hours of mucking around. What I tried first was a USB - TLL adapter set at 3.3V. nothing, Had to change it to 5v then it worked no problem.
>
> Don't use [...]

thank you for the reply. then i will only use rx tx and gnd from the bms.
it seems like i have to use logic converter to drop 5v signal to 3.3v... have you tested if bms accepts 3.3v for its RX pin or it needs the signal to be boosted to 5v as well?

lorneb, 2020-04-20:
My version of the BMS requires a voltage of 5V to work,  3.3 Volts did not work.
The Bluetooth adapter that I bought was 3.3 V and it did not work.  The USB to Serial adapter I have has to voltage settings on it too.  it needed to be set to 5V in order to work with BMS.  So I am 100% sure my BMS only works with 5V

orhanyor, 2020-04-20:
to my understanding it cannot register 3.3v as high signal so it doesnt work. i was asking because some chips have 5v logic but they still accept 3.3v as an acceptable voltage for the inputs. in this case for both rx and tx logic level converter is required. thanks for clearing that up!

closed 2020-04-20 by orhanyor