bres55/Smart-BMS-arduino-Reader #15: discharge current not correct

BreitiDE, 2021-05-05:
hi,
I tested your code and find it very usefull. I found a bug that current is nor shown correctly while discharging. 

Current = 654.97

Charding current is shown correctly. Any chance to fix ... getting the vendorname also would be great :)

Thank you ...

Stefan

bres55, 2021-05-10:
Hi Stefan.
I've been away a while so didn't pick this up.
The code as is, has worked for me, but someone else had a similar problem to you. Can you check this out and see how you get on.
https://github.com/bres55/Smart-BMS-arduino-Reader/issues/5

The vendorname is possible, as I was able to read and write to the Barcode, when I first played around with the BMS, but I am afraid I don't have time to impliment it for you..
I have had my code running faultlessly, for over 2 years now, with a PC picking up the code from the arduino, and so I am able to check the batteries 24/7. It's in a solar panel, battery back up system, the arduino also monitors the charge controller too.

bres55, 2021-07-28:
No one has responded

closed 2021-07-28 by bres55